nucleron/RTE #1: Can it be used with Stm32f103c8t6?

pennybuster, 2018-01-21:
Hi! Thank you for sharing your work!!!

nucleron, 2018-01-22:
Hi!

Currently one of our devices is based on stm32f105xxxx MCU, 
so we have some common STM32F1xx files.

But we have not used Stm32f103c8t6 yet.  
So, you will have to add some BSP files to make RTE work on f103xxxc device.

pennybuster, 2018-01-22:
Thanks!!! It kicks Codesys in the ass! Are you considering porting OSCAT library (http://www.oscat.de/)?

nucleron, 2018-01-23:
>OSCAT
Thanks for the tip!
I'll contact Beremiz people, they may be interested in adding the lib to Beremiz...

pennybuster, 2018-01-23:
Great!!!

closed 2018-01-23 by pennybuster

nucleron, 2018-01-24:
>Great!!!
Actually, the situation is not so great. 

There are some [problems](http://beremiz-devel.2374573.n4.nabble.com/Beremiz-devel-OSCAT-library-td357.html) with OSCAT.

So we need someone to help with that lib...