nucleron/RTE #4: Проблемы с COM-портом.

elSapito, 2018-08-22:
Добрый день!

Пытаюсь подключится из YAPLC-IDE к YAPLC-RTE, прошитой в STM32F4-discovery, не получается. Дошел до отладки связи по COM-порту.
На моем экземпляре discovery, похоже, убит uart1, используемый в RTE. 
В RTE/src/bsp/stm32f4/plc_config.h  меняю настройки DBG_USART на USART2, порты и прерывания, соответственно, тоже.
Тестирую Tx/Rx при помощи терминала на ПК. 
Модифицирую программу RTE для циклической отправки байт, в терминале вижу нужные символы.
Тестирую Rx. Ставлю точку останова в обработчике прерывания USART, ввожу в терминале разные символы. Прерывание срабатывает не на все символы, например, "U" и "u" не работают, "Y" и "y" работают. В отладчике вижу принимаемые байты, они не соответствуют кодам символов, чаще всего это 0xF3, 0xF5, 0xFC. 

Протестировал связь по COM-порту аналогичным образом на проекте из STM32CubeMX, там все прекрасно работает, т.е. проблемы с железом можно исключить.

Подскажите пожалуйста, в чем может быть дело?

nucleron, 2018-08-22:
В режиме работы ком порта, четности и т.д.

elSapito, 2018-08-22:
Настройки uart как в оригинальном проекте, настройки COM-порта на ПК совпадают с ним.
Плюс, на передачу uart в RTE работает нормально.