nucleron/RTE #5: Проблема с отладкой

vikonik, 2019-11-28:
Здравствуйте.
Проект собрался после того как в свойствах компилятора в проекте поменял значение -mcpu-ab на soft, также в библиотеке libopencm3\lib\stm32\F4  в maikfile  FP_FLAGS = -mfloat-abi=soft. проект собрался и запустисля под отладчиком. 
Возниккла такая проблема, программа постоянно перезапускается после попытки установить параметры systick в строке 135 файла plc_tick.c 
Как с этим бороться, Вы можете мне помочь?
В код программы никаких изменений не вносил.

nucleron, 2019-11-29:
Нужно поставить точку останова в plc_irq_stub и отладить предположительно хардфолт.

vikonik, 2019-12-03:
Программа до plc_irq_stub не доходит. Эта точка останова не срабатывает.
Поясните пожалуйста, что Вы понимаете под "отладить предположительно хардфолт"

nucleron, 2019-12-03:
>Поясните пожалуйста, что Вы понимаете под "отладить предположительно хардфолт"
https://community.st.com/s/question/0D50X00009sW72oSAC/how-to-debug-a-hard-fault-exception-for-stm32

>Программа до plc_irq_stub не доходит. Эта точка останова не срабатывает.
Тогда надо поставить точку останова в начало scb_reset_system() и смотреть по трейсбеку, откуда его вызвали.

ALTracer, 2021-07-05:
@vikonik Проще использовать в полную силу ядро Cortex-M4F и вернуться на `-mfloat-abi=hard -mfpu=fpv4-sp-d16`, показав линкеру правильную libgcc.a в новых тулчейнах.